rei-vilo/PDLS_EXT3_Basic_Global #4: Would it be possible to document the API of the EXT3 board?

matyasf, 2022-02-21:
**Is your feature request related to a problem? Please describe.**
I'm trying to get the `Common_WhoAmI` code working on a Raspberry Pi Zero 2, but could not do it even after solving many roadblocks. The main issues are with [RasPiArduino](https://github.com/me-no-dev/RasPiArduino), I guess mainly because the project seems abandoned. The issues that I encountered:
- RasPiArduino required that I enable the root user on the Pi (and SSH login), turn off SPI and i2c kernel modules, turn off sound feature, these are settings that I do not want to do on a Pi.
- I could only get this library running with several workarounds: Needed to solve https://github.com/me-no-dev/RasPiArduino/issues/123 , I needed to include the library code with the project because Arduino's library manager refused to add it.
- Finally I could not copy the compiled code to the Pi because Arduino was trying to use my SSH key for some reason to upload stuff instead of a simple password authentication.
- When I copied it manually the application exited with the very unhelpful `UNKNOWN_REVISION: 0x00902120, MODEL: 0xFFFFFFFF` message.

**Describe the solution you'd like**
A simple document on how to connect and send data to the display would be super useful. I'm not a C++ programmer and would prefer to write the code in other languages, e.g. Python. Something would allow me to write code like this: https://github.com/f-necas/EpaperDisplay/blob/master/epd.py (this is for EXT2)

**Describe alternatives you've considered**
reverse engineering the protocol based on this library

rei-vilo, 2022-02-21:
I am afraid all the issues are related to the [RasPiArduino](https://github.com/me-no-dev/RasPiArduino) core and not to the library.

As answered at https://github.com/PervasiveDisplays/EPD_Driver_GU_mid/issues/1#issuecomment-1046391934, please refer to https://www.pervasivedisplays.com/wp-content/uploads/2019/09/ApplicationNote_EPD740_Mono_with_IST7136_210323.pdf

> You may find more information on the SPI process in this [document](https://www.pervasivedisplays.com/wp-content/uploads/2019/09/ApplicationNote_EPD740_Mono_with_IST7136_210323.pdf). The document is for a 7.4" display, but the process is similar for a 5.8".

matyasf, 2022-02-21:
Thanks for the answer. For future reference I did some investigation:
The `UNKNOWN REVISION` error is coming from [RasPiArduino](https://github.com/me-no-dev/RasPiArduino/blob/master/cores/piduino/wiring.c#L172), the revision number its complaining about is the revision of my Pi Zero 2 W -- I guess it's not supported by RasPiArduino

closed 2022-02-21 by matyasf

rei-vilo, 2022-02-21:
You're welcome! 

You can explore the [BCM2835 library](https://www.airspayce.com/mikem/bcm2835/) for a native Linux application on the Raspberry Pi. It is easy to interface with a C++ program.

matyasf, 2022-02-22:
Thanks! Have you got some example for using the BCM2835 library? I assume that if I make a C++ program where I add the BCM2835 lib and this one as a dependency the build would fail since this library needs `Arduino.h` and `SPI.h` which are Arduino-specific (so I cannot call for example `SPI.setClockDivider()`.

rei-vilo, 2022-02-23:
Actually, the BCM2835 library allows you to develop a native Linux application for the Raspberry Pi. 

I am using the remote extension for Visual Studio Code as described at [Coding on Raspberry Pi remotely with Visual Studio Code](https://www.raspberrypi.com/news/coding-on-raspberry-pi-remotely-with-visual-studio-code/).

matyasf, 2022-02-23:
Yep, I'm using VS Code that way already. My question is rather that am I understanding correctly that for a native app I'd need to rewrite your code with the following:
- Remove `Arduino.h`, `SPI.h` includes, since these are Arduino SDK specific and won't compile on a Pi.
- include `bcm2835.h`
- Rewrite functions that were in `Arduino.h`, `SPI.h` to be `bcm2835.h` specific, for example use `bcm2835_spi_setClockDivider` https://www.airspayce.com/mikem/bcm2835/group__spi.html#gad623657d9513faa89d22ae1dc87eec38 instead of `SPI.setClockDivider()`

Sorry, I have very little C/C++ experience

rei-vilo, 2022-02-27:
You get the general idea for the BCM2835 library. 

Now, I faced no major issues with the [RasPiArduino](https://github.com/me-no-dev/RasPiArduino/blob/master/cores/piduino/wiring.c#L172) port.