rei-vilo/PDLS_EXT3_Basic_Global #6: Cannot complie on Arduino ide 1.8.19

ArthurZhou, 2022-10-01:
**Describe the bug**
cannot complie because of an error

**To Reproduce**
Steps to reproduce the behavior:
1. Run demo: `Common_Coulors`
2. Press `Complie`
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Macos 10.13.6
 - IDE Arduino IDE 1.8.19
 - Version [e.g. 1.8.6]

**Boards (please complete the following information):**
blank

**Additional context**
error log:
```

libraries/PDLS_EXT3_Basic-main/hV_Utilities.cpp.o:(.text._Z12formatStringPKcz+0x4): undefined reference to `vsprintf'
libraries/PDLS_EXT3_Basic-main/hV_Utilities.cpp.o: In function `formatString(char const*, ...)':
/Users/***/Documents/Arduino/libraries/PDLS_EXT3_Basic-main/src/hV_Utilities.cpp:278: undefined reference to `vsprintf'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

rei-vilo, 2022-10-01:
Which board package are you using? With **ESP8266 Community 3.0.2**, compilation succeeds.

[image]

`vsprintf` is defined at /Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/lib/gcc/xtensa-lx106-elf/10.3.0/include/ssp/stdio.h line 61. 

[image]

ArthurZhou, 2022-10-02:
My driver version is 2.6.4
because 3.0.2 cannot been installed correctly on it.

Maybe I should try again on windows.However, thanks for your help

rei-vilo, 2022-10-02:
You're welcome!

closed 2022-10-02 by rei-vilo