rei-vilo/PDLS_EXT3_Basic_Global #13: flushPartial Legacy Version 5

siochs, 2024-02-15:
### Describe the request

I like your library. According to the  [official documentation](https://pdls.pervasivedisplays.com/userguide/Volume3/Chapter3/Section2/#use) there is `partialUpdate()` available in the Legacy Version 5 of this library. I searched in old commits, but I cannot find any implementation. Can you help me out with that?

I am hobbyist and trying to recycle an old shelf label, so no commercial interests in the game here.

### Describe the current behavior

Partial Update routines not available in legacy commits (version 5.x upwards)

### PDLS variant

PDLS_EXT3_Basic_Global

### PDLS version

5.5.0

### IDE or SDK name and version

Arduino SDK, in Platformio

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/vilo/PDLS_EXT3_Basic_Global/issues?q=)
- [X] My request contains all necessary details

rei-vilo, 2024-02-15:
Hi!

Thank you for your kind words.

The partial update mode is deprecated and no longer supported. The Basic edition didn't include that feature.

I would recommend to use fast update instead, available at [PDLS_EXT3_Basic_Fast](https://github.com/rei-vilo/PDLS_EXT3_Basic_Fast). 

Fast update is marginally slower than partial update but ensures an optimal image quality.

siochs, 2024-02-15:
Thank you, that clarifys it.

closed 2024-02-15 by siochs