rei-vilo/PDLS_EXT3_Basic_Global #16: Updating fast update screen with global routines.

yegvla, 2024-04-13:
### Describe the problem

I have acquired some EPDs (E2213KS0E1) and have designed a custom driving circuit based on the "EPD Driving Circuit" document.

Because of the projects structure, I am unable to use the driver you have provided, so I have developed a custom driver to accompany the hardware.  But I am unable to get the global update working on my panel, it sometimes works but there are artifacts visible on the screen:

[image]
This is the resulting image if I send all zeros to DTM1 & DTM2.  Very similar artifacts can be observed when sending an actual image to DTM1 instead of just zeros.

When I do 2 fast updates instead, the image is clear and no artifacts are visible.
DTM1 -> white; DTM2 -> black
DTM1 -> black;  DTM2 -> white
(DTM1 -> white; DTM2 -> image to display)
[image]

Is this normal behavior? Can I only update the screen using the fast update routines or is there some problem on my side?

Sorry for the inconvenience and thanks in advance for your support.

### To reproduce

/

### Expected behavior

/

### IDE or SDK name and version

custom

### PDLS variant

PDLS_EXT3_Basic_Global

### PDLS version

/

### MCU board

nordic nRF52840 (custom)

### Extension board

Custom

### EPD panel reference

CE2213KS0E1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/rei-vilo/PDLS_EXT3_Basic_Global/issues?q=)
- [X] My report contains all necessary details

rei-vilo, 2024-04-13:
Please note:

- There are two series of libraries: **drivers** like [EPD_Driver_GF_small](https://github.com/PervasiveDisplays/EPD_Driver_GF_small) and **libraries** like [PDLS_EXT3_Basic_Fast](https://github.com/PervasiveDisplays/PDLS_EXT3_Basic_Fast).

- If your are referring to the [EPD_Driver_GF_small](https://github.com/PervasiveDisplays/EPD_Driver_GF_small) not to the  [PDLS_EXT3_Basic_Fast](https://github.com/PervasiveDisplays/PDLS_EXT3_Basic_Fast), please open a ticket at [Pervasive Displays Technical Support](https://www.pervasivedisplays.com/technical-support/).

- PDLS_EXT3_Basic_Fast does not support global update, only fast update.

- Fast update and global update are exclusive, as they manage `DTM1` and `DTM2` in different ways. `DTM1` is **new** image in global update but **previous** image in fast update. See section **5. Input image to the EPD** of the [Application Note for small sizes of wide temperature EPDs](https://www.pervasivedisplays.com/wp-content/uploads/2023/02/ApplicationNote_Small_Size_wide-Temperature_EPD_v03_20231031_B.pdf) at [2.13″ E-ink displays - Wide temperature + Fast update](https://www.pervasivedisplays.com/product/2-13-e-ink-displays/).

Now, as there are hardware and software modifications, I would recommend a step-by-step approach detailed in the [Migrate to another SDK](https://pdls.pervasivedisplays.com/userguide/Volume9/Chapter1/Section17/) application note.

1. Test the E2213KS0E1 with EXT3-1 and PDLS_EXT3_Basic_Fast. 

Point 1 ensures the standard configuration works fine.

2. Test the E2213KS0E1 with the custom board and PDLS_EXT3_Basic_Fast. 

Please refer to [Board configuration - Non-listed boards](https://pdls.pervasivedisplays.com/userguide/Volume3/Chapter3/Section2/#board-configuration) to adapt PDLS_EXT3_Basic_Fast to the custom board.

Point 2 ensures the custom board works fine.

3. Test the E2213KS0E1 with EXT3-1 and the custom driver. 

Point 3 ensures the custom board works fine.

Use a logic analyser to record points 1 and 3 and compare the traces to find the issue.
 
4. Test the E2213KS0E1 with the custom board and the custom driver.

yegvla, 2024-04-13:
Thank you for your fast response.

My implementation of PDLS_EXT3_Basic_Fast works as I expect.

I just wanted to clarify if I can also use my display using PDLS_EXT3_Basic_Global, because from my testing it did not seem to work, but if it should be possible then there is probably something wrong in my implementation.

> Fast update and global update are exclusive

So a fast update display can not be updated using the global library?

rei-vilo, 2024-04-13:
The Basic edition of PDLS tries to keep things simple and provide one single update mode for each family. 

- PDLS_EXT3_Basic_Fast only provides fast update for screens with embedded fast update (PS film) and wide temperature (KS film),
- PDLS_EXT3_Basic_Global only provides global update for screens monochrome (CS film) and red (JS film).

Screens with embedded fast update (PS film) and wide temperature (KS film) do support both global and fast update, but PDLS_EXT3_Basic_Fast does not implemented global update.

If you want to implement it, please refer to section **4. Input initial command** of the [Application Note for small sizes of wide temperature EPDs](https://www.pervasivedisplays.com/wp-content/uploads/2023/02/ApplicationNote_Small_Size_wide-Temperature_EPD_v03_20231031_B.pdf).

yegvla, 2024-04-14:
Hello, I tested some more and I have come to the conclusion that some of my EPDs were damaged!
It took me so long to realize because sometimes they manage to refresh fine.

The code worked perfectly fine on a different unit.

Thank you for your guidance and help in resolving my problem.

closed 2024-04-14 by yegvla

rei-vilo, 2024-04-14:
You are welcome. Have a great weekend!