rei-vilo/PDLS_EXT3_Basic_Global #18: Compilation Error: Duplicate case value

eggsactly, 2024-05-16:
### Describe the problem

When attempting to build example Common_Text, user is confronted with this compiler error: 
```
/home/user/Arduino/libraries/PDLS_EXT3_Basic_Global/src/hV_HAL_Peripherals.cpp: In function 'void hV_HAL_Wire_end()':
/home/user/Arduino/libraries/PDLS_EXT3_Basic_Global/src/hV_HAL_Peripherals.cpp:131:10: error: 'class TwoWire' has no member named 'end'
     Wire.end();
```

### To reproduce

1. File > Examples > PDLS_EXT3_Basic_Global > Common_Fonts
2. Comment out line 41 to evade compiler error `Common_Text:41: error: redefinition of 'Screen_EPD_EXT3 myScreen'`
3. Sketch > Verify/Compile

### Expected behavior

IDE should report `Done compiling.`

### IDE or SDK name and version

Arduino IDE 1.8.5

### PDLS variant

PDLS_EXT3_Basic_Global

### PDLS version

8.0.1

### MCU board

Generic ESP8266 Module

### Extension board

EXT3-1

### EPD panel reference

E2266QS0F1

### Additional context

Compilation works on version 7.0.4 of library. 

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/rei-vilo/PDLS_EXT3_Basic_Global/issues?q=)
- [X] My report contains all necessary details

rei-vilo, 2024-05-16:
Hi!

That is strange, as `end()` is part of the `Wire` library.

See [Arduino](https://www.arduino.cc) > [Reference](https://www.arduino.cc/reference/en) > [Language](https://www.arduino.cc/reference/en#language) > [Functions](https://www.arduino.cc/reference/en#functions) > [Communication](https://www.arduino.cc/reference/en#communication) > [Wire](https://www.arduino.cc/reference/en#wire) > [end()](https://www.arduino.cc/reference/en/language/functions/communication/wire/end/)

Please try this minimal code:

```
#include "Wire.h"

void setup()
{
    Wire.begin();
    Wire.end();
}

void loop()
{
    delay(1000);
}

```

eggsactly, 2024-05-16:
I agree, it is strange. The compiler error is coming from _inside_ the library. Is this reproducible? Or is this a "only works (or doesn't) on my machine?

rei-vilo, 2024-05-16:
The [ESP8266 Wire](https://github.com/esp8266/Arduino/blob/master/libraries/Wire/Wire.h) library does not provides the end function. 

You may add it manually.

For questions related to the ESP8266, please have a look at www.esp8266.com or www.esp32.com.

I don't use the ESP8266 anymore because of these nasty surprises.

closed 2024-05-16 by rei-vilo

rei-vilo, 2024-05-21:
Other platforms lack `TwoWire::end()` as well

- Energia for Texas Instruments LM4F/TM4C
- Energia for Texas Instruments CC3200

Because Energia is no longer maintained, support for the Texas Instruments boards is deprecated.