rei-vilo/PDLS_EXT3_Basic_Global #19: panelCS, panelDC not getting set as GPIOs on esp32-s3

ra0943-VK3ACH, 2024-08-03:
### Describe the problem

When running Example_File_Line.ino on an esp32-s3 with custom pin definitions i keep getting the error "[  1585][E][esp32-hal-gpio.c:176] __digitalWrite(): IO 13 is not set as GPIO.
[  1592][E][esp32-hal-gpio.c:176] __digitalWrite(): IO 11 is not set as GPIO." pin 11 is CS and pin 13 is DC 

### To reproduce

Run example_fast_line with the custom board configuration
```
const pins_t myBoard =
{
    .panelBusy = 15, ///< EXT3 and EXT3-1 pin 3 Red
    .panelDC = 13, ///< EXT3 and EXT3-1 pin 4 Orange
    .panelReset = 16, ///< EXT3 and EXT3-1 pin 5 Yellow
    .flashCS = NOT_CONNECTED, ///< EXT3 and EXT3-1 pin 8 Violet
    .panelCS = 11, ///< EXT3 and EXT3-1 pin 9 Grey
    .panelCSS = NOT_CONNECTED, ///< EXT3 and EXT3-1 pin 12 Grey2
    .flashCSS = NOT_CONNECTED, ///< EXT3 and EXT3-1 pin 20 Black2
    .touchInt = NOT_CONNECTED, ///< EXT3-Touch pin 3 Red
    .touchReset = NOT_CONNECTED, ///< EXT3-Touch pin 4 Orange
    .panelPower = NOT_CONNECTED, ///< Optional power circuit
    .cardCS = NOT_CONNECTED, ///< Separate SD-card board
    .cardDetect = NOT_CONNECTED ///< Separate SD-card board
};
```

### Expected behavior

No errors and an output on the display

### IDE or SDK name and version

Platformio - Arduino 3.0.3

### PDLS variant

PDLS_EXT3_Basic_Fast

### PDLS version

8.0.4

### MCU board

ESP32-S3

### Extension board

Custom

### EPD panel reference

SE2152KS0J1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/rei-vilo/PDLS_EXT3_Basic_Global/issues?q=)
- [X] My report contains all necessary details

rei-vilo, 2024-08-03:
You would need to define the pins number for your custom board. Please refer to the documentation of your board.

ra0943-VK3ACH, 2024-08-03:
Hello the pins are i am defining the pins via this constructor
```
const pins_t myBoard =
{
    .panelBusy = 15, ///< EXT3 and EXT3-1 pin 3 Red
    .panelDC = 13, ///< EXT3 and EXT3-1 pin 4 Orange
    .panelReset = 16, ///< EXT3 and EXT3-1 pin 5 Yellow
    .flashCS = NOT_CONNECTED, ///< EXT3 and EXT3-1 pin 8 Violet
    .panelCS = 11, ///< EXT3 and EXT3-1 pin 9 Grey
    .panelCSS = NOT_CONNECTED, ///< EXT3 and EXT3-1 pin 12 Grey2
    .flashCSS = NOT_CONNECTED, ///< EXT3 and EXT3-1 pin 20 Black2
    .touchInt = NOT_CONNECTED, ///< EXT3-Touch pin 3 Red
    .touchReset = NOT_CONNECTED, ///< EXT3-Touch pin 4 Orange
    .panelPower = NOT_CONNECTED, ///< Optional power circuit
    .cardCS = NOT_CONNECTED, ///< Separate SD-card board
    .cardDetect = NOT_CONNECTED ///< Separate SD-card board
};
```
The problem is when running the code the error "[  3584][E][esp32-hal-gpio.c:176] __digitalWrite(): IO 13 is not set as GPIO." and 
"[  3591][E][esp32-hal-gpio.c:176] __digitalWrite(): IO 11 is not set as GPIO."  appears and keep repeating

rei-vilo, 2024-08-03:
Have you tried on a separate program

```
pinMode(13, OUTPUT);
```

and same for 11?

Again, you need to refer to the documentation of your board to select the right pins numbers.

ra0943-VK3ACH, 2024-08-03:
pinMode and digital write works on another program, I think I have found the problem though. When calling ```hV_HAL_SPI3_define(12,10)``` to change the spi pins to 12 and 10 respectively this does not appear to change the SPI pins. they remain as the default 11,12,13 which I would prefer not to use for ease PCB design.

rei-vilo, 2024-08-03:
Glad to know you found the issue. 

SPI management by the ESP32 core for Arduino is a bit difficult, as it sometimes the pins sued differ from the macros SCK and MOSI. 

The SPI3 has a specific definition for ESP32-C3 but indeed, ESP32-S3 was considered as standard. Maybe it should have its own section too.

https://github.com/rei-vilo/PDLS_EXT3_Basic_Global/blob/b7b9d61f71ce9c661b541d3617195f461efe1f2d/src/hV_HAL_Peripherals.cpp#L237-L252

ra0943-VK3ACH, 2024-08-03:
I have fixed errors with pins not being defined correctly, however, still nothing is appearing on the display. When looking at the spi signals, it appears that the clock line is pulsing in groups of 2 or 3 (seemingly random but more often 3) and mosi pulses do not line up with the clock rising or falling edge 100% of the time

closed 2024-08-03 by ra0943-VK3ACH