rei-vilo/PDLS_EXT3_Basic_Global #20: Unable to run fast update example on E2417PS0D1 screen with Arduino Nano Matter

lucasvickers, 2024-08-28:
### Describe the problem

I'm unable to execute the eink-example-fastline.ino example for the following combination

- E2417PS0D1
- Arduino Nano Matter

When running the example, execution stops at this line: https://github.com/PervasiveDisplays/PDLS_EXT3_Basic_Fast/blob/main/src/Screen_EPD_EXT3.cpp#L1098

It appears to allocate a uint8_t[30,000] array. If I change the value to something arbitrarily low like 10, it passes this line and fails elsewhere (expected).

Strangely, if I add `new uint8_t[30000];` at the beginning of my sketch, it does not fail.

I'm unable to get further details out of the debugger (could be limited knowledge on my part).
I'm also seeing this issue with and without the eink screen connected

### To reproduce

Run eink-example-fastline example with:
- E2417PS0D1
- Arduino Nano Matter
- Arduino IDE 2.3.2
- PDLS_EXT3_Basic_Fast 8.0.4
- EXT3-1 Dev Board

### Expected behavior

It runs the demo code as expected

### IDE or SDK name and version

2.3.2

### PDLS variant

PDLS_EXT3_Basic_Fast

### PDLS version

8.0.4

### MCU board

Arduino Nano Matter

### Extension board

EXT3-1

### EPD panel reference

SE2417PD0D1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/rei-vilo/PDLS_EXT3_Basic_Global/issues?q=)
- [X] My report contains all necessary details

rei-vilo, 2024-08-28:
Thank you for the detailed report. I was able to replicate the issue. Possible causes may include memory leak and memory fragmentation.

I suggest you to use this workaround, if you plan to use the E2417PS0D1 only.

* Comment out the lines

https://github.com/rei-vilo/PDLS_EXT3_Basic_Global/blob/b7b9d61f71ce9c661b541d3617195f461efe1f2d/src/Screen_EPD_EXT3.cpp#L1110

and

https://github.com/rei-vilo/PDLS_EXT3_Basic_Global/blob/b7b9d61f71ce9c661b541d3617195f461efe1f2d/src/Screen_EPD_EXT3.cpp#L1307-L1312

* Replace the line

https://github.com/rei-vilo/PDLS_EXT3_Basic_Global/blob/b7b9d61f71ce9c661b541d3617195f461efe1f2d/src/hV_Screen_Buffer.h#L409

with 

```
uint8_t s_newImage[30000];
```

rei-vilo, 2024-08-29:
Glad it works fine. 

Stay tuned to **Excessive use of RAM by certain static objects** at https://github.com/SiliconLabs/arduino/issues/36 for a solution of the root cause.

closed 2024-08-29 by rei-vilo