rei-vilo/PDLS_EXT3_Basic_Global #22: myScreen.rectangle does not clear text from draw area, as would be expected

lucasvickers, 2024-09-13:
### Describe the problem

I have some [test code](https://gist.github.com/lucasvickers/ac9f33a178b687c56e807d3ae213bc5f) (will explain in detail below) that updates the center area of the screen to understand fast update performance. I use myScreen.rectangle to 'clear' the area I'm working in, and while this seems to work for lines, it does not work for text. I believe this is a bug, or I'm not understanding the API correctly.

Longer explanation. I am exploring a UI where numbers will update on the screen. I want to update these numbers as quickly as possible to demonstrate expected latency from a fast update screen. I only have access to a limited number of fonts, so I am framing the font with alternating lines to make a larger 'working area'. I am doing this as I assume it will increase the region of the screen that is updating and provide me with a more realistic update scenario if I were to use a much larger font.

I could use your input to validate the following:
1. I assume fast update speed is dependent on how much of the screen we are updating, therefore calling [rectangle](https://gist.github.com/lucasvickers/ac9f33a178b687c56e807d3ae213bc5f#file-gistfile1-txt-L48) and [framing](https://gist.github.com/lucasvickers/ac9f33a178b687c56e807d3ae213bc5f#file-gistfile1-txt-L57-L65) my area with lines would be equivalent to using a font that fit inside the framing box. Is this true?
2. I assume drawing a white rect over the working area would clear previous written text, is this true?
3. I'm generally using the API as expected

### To reproduce

Run the following code
https://gist.github.com/lucasvickers/ac9f33a178b687c56e807d3ae213bc5f

### Expected behavior

I would expect the text would update from 15, to 72, 36, 99, with the previous text being removed each time. This is not working.

I would expect the framing lines to alternate. This is working.

I expect the refresh time to be roughly the same as I could expect if I used a font that were large and fit just inside my framing lines.

### IDE or SDK name and version

Arduino IDE 2.3.2

### PDLS variant

PDLS_EXT3_Basic_Fast

### PDLS version

8.0.4 with modifications found in https://github.com/rei-vilo/PDLS_EXT3_Basic_Global/issues/20

### MCU board

Arduino Nano Matter

### Extension board

EXT3-1

### EPD panel reference

SE2417PD0D1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/rei-vilo/PDLS_EXT3_Basic_Global/issues?q=)
- [X] My report contains all necessary details

lucasvickers, 2024-09-13:
I resolved my issue by adding setPenSolid() before rectangle(), tho I'll leave this open to get feedback on the fast refresh latency expectations.

rei-vilo, 2024-09-13:
Answering your questions

> I assume fast update speed is dependent on how much of the screen we are updating, therefore calling [...]

The update speed is fixed, as the whole frame-buffer (previous screen, new screen) is sent to the screen and the whole screen is refreshed.

> I assume drawing a white rect over the working area would clear previous written text, is this true?

You have two options:

1. Draw a solid rectangle and then print the text

```
myScreen.gText(0, 0, "Old");
myScreen.refresh();
delay(5000);

myScreen.setPenSolid(true);
myScreen.rectangle(0, 0, 200, 100);
myScreen.gText(0, 0, "New");
myScreen.refresh();
delay(5000);
```

2. Print the text is solid mode.

```
myScreen.gText(0, 0, stringFormat("%8s", "Old"));
myScreen.refresh();
delay(5000);

myScreen.setFontSolid(true);
myScreen.gText(0, 0, stringFormat("%8s", "New"));
myScreen.refresh();
delay(5000);
```

`formatString()` ensures the correct formatting.

> I'm generally using the API as expected

Please refer to

* `setPenSolid()` [reference](https://pdls.pervasivedisplays.com/reference/html/d4/d94/a00967.html#a4977052c12742168ae102b6ae63adc17) and [example](https://pdls.pervasivedisplays.com/userguide/Volume3/Chapter3/Part2/Shapes/#configure)
* `setFontSolid()` [reference](https://pdls.pervasivedisplays.com/reference/html/d4/d94/a00967.html#a815b912710286e95a915ef30599c3e4e) and [example](https://pdls.pervasivedisplays.com/userguide/Volume3/Chapter3/Part2/Text/#configure)
* `formatString()` [reference](https://pdls.pervasivedisplays.com/reference/html/d6/d80/a00371.html#a78c5901b47f8f565532d68e96d5427ec)

lucasvickers, 2024-09-13:
> The update speed is fixed, as the whole frame-buffer (previous screen, new screen) is sent to the screen and the whole screen is refreshed. This procedure ensures the optimal image quality.

This is more of a e-ink product question, not a library question. Conceptually, would it be possible to only update a subset of the screen to further increase the speed of refresh?

Thanks!

rei-vilo, 2024-09-14:
> Conceptually, would it be possible to only update a subset of the screen to further increase the speed of refresh?

It is more complicated than that: we need to consider selecting and sending the frame-buffer, and then refreshing the screen.

Sending only part of the frame-buffer (previous screen, new screen) instead of the whole frame-buffer could theorically save some milliseconds. 

However, the algorithm on the MCU to select part of the frame-buffer is more complicated and thus relatively slower than the algorithm to send the whole frame-buffer. On small screens, given the high SPI speed, the difference in time is minimal.

The refresh process always includes the whole screen. One major drawback of partial update was ghosting. Fast update ensures optimal image quality.

As a reference, please refer to the wiki page [Updating the Display | PDI Public Wiki (pervasivedisplays.com)](https://docs.pervasivedisplays.com/epd-overview/updating-the-display).

lucasvickers, 2024-09-14:
Got it, this is all really helpful context. Thank you!

closed 2024-09-14 by lucasvickers